emsesp/EMS-ESP32 #2585: Boiler Input 4 resets upon ems-esp reboot

timostr, 2025-06-25:
### DESCRIPTION
I'm using "boiler input 4" to raise the heating water temperature if enough photovoltaic energy is available. This is working fine - Thank you!
I noticed that the ems-esp device reboots every 1-3 days. After the reboot the bit for the photovoltaic is not set anymore.
At the heatpump user interface the setting is still correct (enabled for photovoltaic).
[[](url)](https://photos.app.goo.gl/7RWw3mije2FM1dPc7)
Once I toggle at the heatpump the photovoltaic setting, the "boiler input 4" is again correct.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [X] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
{"system":{"version":"3.7.3-dev.13","uptime":"000+08:46:55.202","uptimeSec":31615,"platform":"ESP32S3","cpuType":"ESP32-S3","arduino":"ESP32 Arduino v2.0.17","sdk":"v4.4.7-dirty","freeMem":195,"maxAlloc":171,"freeCaps":8324,"usedApp":2158,"freeApp":3538,"partition":"app0","flash_chip_size":16384,"resetReason":"Software reset CPU / Software reset CPU","psram":true,"psramSize":8189,"freePsram":8130,"model":"","temperature":43.4},"network":{"network":"WiFi","hostname":"ems-esp","RSSI":-53,"WIFIReconnects":0,"TxPowerSetting":78,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"connected","enabled":true,"server":"fritz.box","tzLabel":"Europe/Berlin","timestamp":1750831563},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":10286,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTReconnects":0,"enabled":true,"clientID":"ems-esp","keepAlive":60,"cleanSession":false,"entityFormat":3,"base":"ems-esp","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":10,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":false},"syslog":{"enabled":false},"sensor":{"temperatureSensors":0,"temperatureSensorReads":0,"temperatureSensorFails":0},"api":{"APICalls":4218,"APIFails":1},"bus":{"busStatus":"connected","busProtocol":"HT3","busTelegramsReceived":123526,"busReads":29310,"busWrites":10,"busIncompleteTelegrams":8,"busReadsFailed":0,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"S32S3","locale":"en","txMode":2,"emsBusID":11,"showerTimer":false,"showerMinDuration":180,"showerAlert":false,"hideLed":false,"noTokenApi":false,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":false,"telnetEnabled":false,"maxWebLogBuffer":25,"modbusEnabled":false,"forceHeatingOff":false,"developerMode":false},"devices":[{"type":"boiler","name":"XCU_THH/CS5800i, CS6800i, WLW176i, WLW186i","deviceID":"0x08","productID":8,"brand":"Bosch","version":"00.00","entities":167,"handlersReceived":"0xBF 0xC6 0x14 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F","handlersFetched":"0xE6 0xEA 0x048D 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D 0x04AE 0x04AF 0x02CC 0x04AA 0x04A7 0x02D6","handlersPending":"0x10 0x11 0xC2 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x048A 0x04A5","handlersIgnored":"0x04C9 0xE7 0x061E 0x023E 0x0240 0x0291 0x0292 0x0293 0x0294 0x029B 0x02A5 0x02F5 0x02CD 0x04A8 0x04A9 0x04AB 0x04AC 0x04CA 0x04CB 0xD2 0xE2 0x17 0x0497 0xF7"},{"type":"thermostat","name":"HMI800.2/Rego 3000, UI800, Logamatic BC400","deviceID":"0x10","productID":253,"brand":"","version":"47.09","entities":78,"handlersReceived":"0x06 0x02BA 0x02BB 0x02BC 0x031D 0x0267 0xBF 0xC0","handlersFetched":"0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E 0x016E","handlersPending":"0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x02F6 0x031B 0x031E 0x0269","handlersIgnored":"0x0507 0x0508 0x0509 0x050A 0x059D 0xF7"},{"type":"mixer","name":"MM100/MM100","deviceID":"0x20","productID":160,"brand":"","version":"24.05","entities":7,"handlersReceived":"0x02D7","handlersFetched":"0x02CD","handlersIgnored":"0x02E1 0x02EB 0x0255 0xE6 0x0291 0xBF"},{"type":"gateway","name":"K30RF/WiFi module","deviceID":"0x48","productID":252,"brand":"Bosch","version":"12.02","entities":0,"handlersIgnored":"0xF9 0x2040 0x0291 0x0292 0x0B10 0x0495 0x0494 0x04AF 0x04AE 0x0DE8 0x0DE9 0x0DEC 0x0DED 0x0DEF 0x0DF0 0x0293 0x0294 0x029B 0x029C 0x029D 0x029E 0x02AF 0x02B0 0x02F5 0x0B11 0x02B1 0x02B2 0x02FF 0x0B12 0xFF07 0x02B9 0x02BA 0x02BB 0x02BC 0x02C3 0x02C4 0x02C5 0x02C6 0x02CC 0x0B15 0xEA 0x02E0 0x0449 0x044A 0x044B 0x044C 0x0471 0x0472 0x0473 0x0474 0x0484 0x0486 0x0B16 0xFE06 0x0B17 0x0B18"}]}

### Systemlog:
`2025-06-24T23:19:09.000 INFO 5: [emsesp] Starting Scheduler service
2025-06-24T23:19:09.000 INFO 6: [emsesp] Starting Custom Entity service
2025-06-24T23:19:09.000 INFO 7: [temperaturesensor] Starting Temperature Sensor service
2025-06-24T23:19:09.000 INFO 8: [emsesp] Starting Modules service
2025-06-24T23:19:09.000 INFO 9: [module] No external library modules found
2025-06-24T23:19:09.000 INFO 10: [emsesp] Starting Web Server
2025-06-24T23:19:12.267 INFO 11: [emsesp] WiFi connected (Local IP=192.168.188.132, hostname=ems-esp, TxPower=19.5 dBm)
2025-06-24T23:19:12.292 INFO 12: [emsesp] Starting mDNS Responder service
2025-06-24T23:19:12.292 INFO 13: [emsesp] Starting NTP service
2025-06-24T23:19:12.354 INFO 14: [mqtt] MQTT connected
2025-06-24T23:19:12.783 WARNING 15: [command] Command failed: unknown device boiler
2025-06-24T23:19:16.780 INFO 16: [system] NTP connected
2025-06-24T23:19:17.882 INFO 17: [emsesp] Detected EMS device: boiler (0x08)
2025-06-24T23:19:18.100 INFO 18: [emsesp] Detected EMS device: thermostat (0x10)
2025-06-24T23:19:18.521 INFO 19: [emsesp] Detected EMS device: mixer (0x20)
2025-06-24T23:19:18.733 INFO 20: [emsesp] Detected EMS device: gateway (0x48)
2025-06-24T23:19:29.001 INFO 21: [command] Called command boiler/hpin4opt (input 4 options) with value 000000100000
2025-06-25T01:34:15.290 WARNING 22: [telegram] Incomplete Rx: 90 48 FF 32 0A 12 00 00 08
2025-06-25T02:34:15.263 WARNING 23: [telegram] Incomplete Rx: 90 48 FF 00 01 C3 03 10 01 54 03 FF 03 FF 03 FF 03 FF 03 10 01 54 03 FF 08
2025-06-25T02:34:17.492 WARNING 24: [telegram] Incomplete Rx: 90 48 FF 00 01 C5 03 18 01 5C 03 FF 01 FF 03 FF 01 08
2025-06-25T02:34:19.001 WARNING 25: [telegram] Incomplete Rx: 90 48 FF 00 08
2025-06-25T05:34:15.070 WARNING 26: [telegram] Incomplete Rx: C8 90 FF 00 19 03 4B 55 08
2025-06-25T05:34:17.684 WARNING 27: [telegram] Incomplete Rx: C8 90 FF 01 02 03 71 0B 08
2025-06-25T06:34:17.470 WARNING 28: [telegram] Incomplete Rx: C8 90 FF 96 19 08
2025-06-25T07:34:15.169 WARNING 29: [telegram] Incomplete Rx: 90 48 FF 4B 0A 18 08
2025-06-25T08:07:41.419 WARNING 30: [telegram] Incomplete Rx: 10 00 90 00 08 00 88 00 20
2025-06-25T08:07:42.602 WARNING 31: [telegram] Incomplete Rx: 10 00 90 00 08 00`
```

### TO REPRODUCE

Here the flow:
1. Check that correct setting are set for "ems-esp Boiler Input 4 options" = 000000100000 at ems-esp web interface
2. ems-esp device reboots
3. Wrong setting at "ems-esp Boiler Input 4 options" = 000000000000
4. Toggle photvoltaic from Yes to No and Yes again at the heatpump user interface
5. Correct setting  again at "ems-esp Boiler Input 4 options" = 000000100000

### EXPECTED BEHAVIOUR

I would expect that:
a) that "ems-esp Boiler Input 4 options" is same as before reboot
b) ems-esp does not spontanous reboots

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

MichaelDvP, 2025-06-25:
b) for the reboots see #2581. You have also mqtt and api active with a lot of messages. You can go back to dev11 or use dev13 from my repo, with asyncTCP 3.4.2. I'm still testing, but seems that running asyncTCP on a different core than arduino(udp) and mqtt helps also.

a) it takes some minutes until ems-esp reads all values from the bus, the inputs are on a long 50 byte telegram and to get all parts will take a while. 
Your log shows that you are writing commands to this telegram in the first minute, that could cause some strange effects.

MichaelDvP, 2025-06-25:
Can you set the syslog to all, reboot and attach the log of the first 5 minutes. 
If the option4 shows wrong states, try in web system/log or terminal: read '8 486' and check if after the values are correct.
BTW: option4 should show only110 bytes, option 1-3 show 12 bytes.

MichaelDvP, 2025-06-25:
Oh, these new bosch heatpumps break with every rule, requesting 25 bytes and reply with 23, so we expect end of telegram, but the input 4 offsets are in offset 40... 
```
2025-06-25T15:22:44.336279+00:00 ems-esp emsesp 000+00:01:02.542 T 400: me(0x0B) R boiler(0x08), HpInConfig(0x0486), length: 25
2025-06-25T15:22:44.436401+00:00 ems-esp emsesp 000+00:01:02.578 T 401: boiler(0x08) W me(0x0B), HpInConfig(0x0486), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
Difficult to solve. For testing https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

closed 2025-06-29 by timostr