emsesp/EMS-ESP32 #2590: Discrepancy room temperature / RF room temperature

quark654, 2025-07-01:
Hi everyone,

I'm using a Buderus burner GB125/35 to which an ems-esp module is attached. The required and current room temperature is read from the thermostatic valves and sent to the module using mqtt. There are 3 heating circuits, of which 2 are controlled this way (hc1/hc3) using the remotetemp and seltemp commands.
This has been working fine for quite some time, up until recently when I noticed the hc3 current temperature is not the one from the RF remote but rather the one from the hc2 heating circuit. For clarity I added the temperature logs below. As you can see, since the 13th of June, the hc3 room temperature and RF room temperature started deviating. This is a problem because the burner is turned on/off based on the room temperature, not the RF room temperature.

[image]

Meanwhile I tried rebooting, updating to the latest stable and also the latest development release, but alas to no avail. I also tried switching back and forth the hc3 control device, but also no luck. I hope any of you might have a clue of what's going on.

Thanks in advance!

VlastiBroucek, 2025-07-02:
IMHO, if it did not start with the update of the EMS-ESP software, which 

> Meanwhile I tried rebooting, updating to the latest stable and also the latest development release, but alas to no avail.

suggests, then it must be change in your system or HA, not issue with EMS-ESP.

quark654, 2025-07-03:
I believe you're right. I reset the main RC310 thermostat and reconfigured it as it was, for now everything seems to be working as expected. I'll close this issue until I have more information.

closed 2025-07-03 by quark654