emsesp/EMS-ESP32 #2609: setting dhw max power does not work for Junkers ZBS14-5C23 on EMS-ESP

eh-gh, 2025-08-17:
### DESCRIPTION

setting "dhw max power" on EMS-ESP does not work for Junkers ZBS14-5C23
checking with the boiler the followings identified:

dhw max 75%: boiler(0x08) B all(0x00), ?(0x27), data: 00 00 01 00 00 00 00 00 00 00 4B 28 3C 64
dhw max 50%: boiler(0x08) B all(0x00), ?(0x27), data: 00 00 01 00 00 00 00 00 00 00 32 28 3C 64

offset 10 stores the correct value

ems-esp:# read 8 27 10 1
2025-08-17T10:16:06.434 N 484: [emsesp] boiler(0x08) W me(0x0B), ?(0x27), data: 4B (offset 10)

### REQUESTED INFORMATION

{"system":{"version":"3.7.3-test.15","uptime":"006+13:50:32.913","uptimeSec":568232,"platform":"ESP32","cpuType":"ESP32-D0WD-V3","arduino":"Tasmota Arduino v2.0.17","sdk":"4.4.8.240628","freeMem":148,"maxAlloc":91,"freeCaps":100,"usedApp":1675,"freeApp":309,"partition":"app1","flash_chip_size":4096,"resetReason":"Software reset CPU / Software reset CPU","psram":false,"model":""},"network":{"network":"WiFi","hostname":"ems-esp","RSSI":-74,"WIFIReconnects":0,"BSSID":"set","TxPowerSetting":78,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"disconnected","enabled":true,"server":"at.pool.ntp.org","tzLabel":"Europe/Vienna","timestamp":1755418120},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":319634,"MQTTQueued":0,"MQTTPublishFails":2,"MQTTReconnects":2,"enabled":true,"clientID":"ems-esp","keepAlive":60,"cleanSession":false,"entityFormat":4,"base":"xxx","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":10,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":false},"syslog":{"enabled":false},"sensor":{"temperatureSensors":0,"temperatureSensorReads":0,"temperatureSensorFails":0,"analogSensors":2,"analogSensorReads":2,"analogSensorFails":0},"api":{"APICalls":64,"APIFails":0},"bus":{"busStatus":"connected","busProtocol":"HT3","busTelegramsReceived":625375,"busReads":160901,"busWrites":266,"busIncompleteTelegrams":5,"busReadsFailed":1,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"NODEMCU","locale":"en","txMode":3,"emsBusID":11,"showerTimer":false,"showerMinDuration":180,"showerAlert":false,"hideLed":false,"noTokenApi":true,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":6,"boolDashboard":1,"enumFormat":1,"analogEnabled":true,"telnetEnabled":true,"maxWebLogBuffer":25,"modbusEnabled":false,"forceHeatingOff":false,"developerMode":true},"devices":[{"type":"boiler","name":"CBS/GBx72, Trendline, Cerapur, Greenstar Si","deviceID":"0x08","productID":123,"brand":"","version":"06.03","entities":81,"handlersReceived":"0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04","handlersFetched":"0x14 0x16 0x33 0x26 0x28","handlersPending":"0xBF 0xC2 0xC6 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x2E 0x3B","handlersIgnored":"0x23 0x27"},{"type":"thermostat","name":"FW200","deviceID":"0x10","productID":106,"brand":"Junkers","version":"12.14","entities":34,"handlersReceived":"0x06 0x0166","handlersFetched":"0x016F 0x0165 0xBB 0x01FB 0x01D3","handlersPending":"0xA3 0xA2 0x12 0x13 0x0170 0x0171 0x0172 0x022C 0x022D 0x022E 0x022F 0x0230 0x0231 0x0233 0x0234 0x0235 0x0236 0x0237 0x0238 0x0167 0x0239 0x023A 0x023B 0x023C 0x023D 0x023E 0x0168 0x023F 0x0240 0x0241 0x0242 0x0243 0x0244 0x23 0x020F","handlersIgnored":"0xBE"},{"type":"controller","name":"Controller","deviceID":"0x09","productID":152,"brand":"","version":"03.01","entities":0,"handlersIgnored":"0xF9 0x18 0x25 0x04 0x16 0x1C 0x33 0x19 0x28 0x05 0x34 0x26 0x27 0x15 0xF8 0x2A"},{"type":"analogsensor","name":"analogsensor","entities":2},{"type":"scheduler","name":"scheduler","entities":3},{"type":"custom","name":"custom","entities":7}]}

MichaelDvP, 2025-08-17:
Hmm, wwMaxPower was reported here https://github.com/emsesp/EMS-ESP/issues/740#issuecomment-796673687
as telegram 0x26, offset 7. Please:
- send the response of `read 8 26`
- create a custom entitiy for telegram 0x27, offset 10, UINT8, writeable, and check if this works.

Telegram 0x27 is not known, any idea what the other values are? offsets 11, 12, 13 looks like 40°C, 60°C, 100%, maybe limits for temperature and pump?

eh-gh, 2025-08-17:
ems-esp:# read 8 26
2025-08-17T13:09:18.699 N 857: [emsesp] boiler(0x08) W me(0x0B), UBASettingsWW(0x26), data: 01 04 00 0E 00 1E 46 4B
ems-esp:# read 8 27
2025-08-17T13:09:30.905 N 858: [emsesp] boiler(0x08) W me(0x0B), ?(0x27), data: 00 00 01 00 00 00 00 00 00 00 32 28 3C 64

custom entity works:

[image]

closed 2025-09-27 by github-actions[bot]