emsesp/EMS-ESP32 #2636: Add support for Buderus SRC Plus connected room thermostats

Roxxor91, 2025-09-26:
### DESCRIPTION

Add support for Buderus SRC Plus connected room thermostats. (https://www.buderus.de/de/produkte/neue-produkte/logamatic-src-plus-37184)

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ X] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ X] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ X] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ X] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  Paste System information here....
{"system":{"version":"3.7.3-dev.15","uptime":"000+02:28:42.111","uptimeSec":8922,"platform":"ESP32S3","cpuType":"ESP32-S3","arduino":"ESP32 Arduino v2.0.17","sdk":"v4.4.7-dirty","freeMem":187,"maxAlloc":163,"freeCaps":8311,"usedApp":2163,"freeApp":2445,"partition":"boot","flash_chip_size":16384,"resetReason":"Software reset CPU / Software reset CPU","psram":true,"psramSize":8189,"freePsram":8126,"model":"BBQKees Electronics S3 rev.1.1/2025-7-1 (fuse 0x11111971)","temperature":45.8},"network":{"network":"WiFi","hostname":"BBQKeesElectronicsS3","RSSI":-33,"WIFIReconnects":0,"BSSID":"set","TxPowerSetting":0,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"connected","enabled":true,"server":"time.google.com","tzLabel":"Europe/Berlin","timestamp":1758918803},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":1044,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTReconnects":1,"enabled":true,"clientID":"esp32-f3e385a0","keepAlive":60,"cleanSession":false,"entityFormat":3,"base":"ems-esp","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":60,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":false},"syslog":{"enabled":false},"sensor":{"temperatureSensors":0,"temperatureSensorReads":0,"temperatureSensorFails":0,"analogSensors":0,"analogSensorReads":0,"analogSensorFails":0},"api":{"APICalls":0,"APIFails":0},"bus":{"busStatus":"connected","busProtocol":"Buderus","busTelegramsReceived":51542,"busReads":3297,"busWrites":0,"busIncompleteTelegrams":158,"busReadsFailed":0,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"S32S3","locale":"en","txMode":2,"emsBusID":11,"showerTimer":false,"showerMinDuration":180,"showerAlert":false,"hideLed":false,"noTokenApi":false,"readonlyMode":true,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":true,"telnetEnabled":false,"maxWebLogBuffer":50,"modbusEnabled":false,"forceHeatingOff":false,"developerMode":false},"devices":[{"type":"boiler","name":"XCU_THH/CS5800i, CS6800i, WLW176i, WLW186i","deviceID":"0x08","productID":8,"brand":"Buderus","version":"00.00","entities":165,"handlersReceived":"0xBF 0xC6 0x14 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F","handlersFetched":"0xE6 0xEA 0x048D 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D 0x04AE 0x04AF 0x02CC 0x02D6","handlersPending":"0x10 0x11 0xC2 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x048A 0x04A5 0x04AA 0x04A7","handlersIgnored":"0x04C0 0x02E0 0x02EA 0xE7 0x061E 0x0240 0x0291 0x0292 0x0293 0x0294 0x029B 0x02A5 0x04A8 0x04AC 0x02F5 0x04CA 0x04CB 0xF7 0xF6 0x04A0"},{"type":"thermostat","name":"HMI800.2/Rego 3000, UI800, Logamatic BC400","deviceID":"0x10","productID":253,"brand":"","version":"47.07","entities":78,"handlersReceived":"0x06 0x02BA 0x02BB 0x02BC 0x031D 0x0267 0xBF 0xC0","handlersFetched":"0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E 0x016E","handlersPending":"0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x02F6 0x031B 0x031E 0x0269","handlersIgnored":"0x0507 0x0508 0x0509 0x050A 0x059D"},{"type":"heatpump","name":"HP Module","deviceID":"0x38","productID":252,"brand":"","version":"12.02","entities":0,"handlersPending":"0x047B 0x0998 0x0999 0x09A0 0x099B 0x099C 0x04AE 0x04AF 0x099A 0x012E 0x013B","handlersIgnored":"0x0C5B 0x1683 0x0A59 0x0C57 0x167F 0x045D 0x0A61 0x1687 0x0C5F 0x0C59 0x1681 0x0A5A 0x0C58 0x1680 0x0A62 0x0C60 0x1688 0x0C5A 0x1682 0x0C5C 0x1684 0x0A5F 0x0C5D 0x1685 0xBF 0x0C5E 0x1686 0x0C56 0x167E 0x0A57 0x0A58 0x0A5B 0x0A5C 0x0A5D 0x0A5E 0x0A60 0x0A63 0x0A64 0x0A65 0x0A66 0x0C55 0x167D 0x0C61 0x0C62 0x0C63 0x0C64 0x0CAF 0x1689 0x168A 0x168B 0x168C 0x16E1 0x16E2 0x16E3 0x16E4 0x16E5 0x16E6 0x16E7 0x16E8 0x16E9 0x16EA 0x16EB 0x16EC 0x16ED 0x16EE 0x16EF 0x16F0 0x16F5"},{"type":"gateway","name":"MX400/MX400","deviceID":"0x48","productID":17,"brand":"Buderus","version":"12.02","entities":0,"handlersIgnored":"0x2040 0xF9 0x0CBD 0x0CC3 0xF7 0x0CC2 0x0495 0x0494 0x04AF 0x04AE"},{"type":"gateway","name":"MX400/MX400","deviceID":"0x50","productID":17,"brand":"","version":"12.02","entities":0,"handlersIgnored":"0xF7 0xBF 0x0BDD 0x0BDE 0x0BDF 0x0BE0 0x0BE2 0x0BE3 0x0BE4 0x0BE5 0x0BE7 0x0BE8 0x0BE9 0x0BEA 0x0BEC 0x1244 0x1245 0x1246 0x1248 0x1249 0x124A 0x124C 0x124D 0x124E 0x1250 0x1251 0x1252 0x1253 0x1247 0x124B 0x2040 0x124F 0x0BE1 0x0BE6 0x0BEB"}]}

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

Roxxor91, 2025-09-26:
Ok guys. Please dont be mad. I do not intend to spam. It may not be perfect. But I threw the logs and my local readings at an AI. It actually correctly identified actual values from my house in the logs, and I let it create a technical documentation. I will post this here.

Roxxor91, 2025-09-28:
> Funny AI analysis, reads good, looks professional and is mostly wrong ;-)

Funny thing about AI. Because the table in the second AI post is completely correct. But I guess thats the thing: It does it right, but actually doesnt "understand" (as it doesn't "understand" anything) how it got there and writes something.
> You have 14 thermostats, but i can see only 13 in the monitor telegrams 0x0BDD-0x0BEC, where is teh 14th? The AI interpretation with humidity and seltemp ssem right, but actual temp is 16 bit, first 2 bytes. 3 thermostats shows temp 0x8002, this is an error code, we know bosch uses 0x7F00 and 0x8000 for disconnected sensors, This seems something similar. What is different on the last 3 sensors, We see 3 possible states , shown with valid temperature, shown with code and not shown. Please explain the differences in your configuration.

I am terribly sorry. Should have edited my first post. Actually there are 12 Thermostats. I got 14 already used therostats cheaply, but only installed 12. The table "Complete Thermostat Installation Map" by the AI is correct(at least name, humidity, temp and set point).

> The F7 telegram is for configuring the valid settings in the HMI menu. The data shows 3 bytes for the telegram id and one or more bytes as bitfiled for valid offsets. e.g. `gateway(0x50) B all(0x00), MenuConfig(0xF7), data: FF 0A BA D9` means: for the telegram `FF0ABA`, in ems-esp notation `0x0BBA`, the offsets 0, 3, 6, 7 ware valid (D9 has bits 0,3,6,7 set). Changing mode seems to write something to offset 0 of telegrams 0x0BB5..0x0BC4, sadly we have not all mode changes in the log, your fist post shows only 03 for mode off. For mode auto the F7 sets offset 1 as valid, mode manual offset 3, i guess these are the setpoints. Your first post shows writing the setpoint to offset 3 in manual mode.
> 
> Please do in terminal `read 50 BB5` then change the modes for the first thermostat, Also change the setpoint, between each mode-change. Then we should see all settings. There seems to be telegram 0x1230..0x123F, not broadcasted but config is read. Please do a `read 50 1230` Broadcasted for all thermostats are telegrams 0x1244..0x1253, containing for all the same values (`data: 00 00 00 03 00 00 00 00 00 00 00 00 00 00`)

Ok. I used "BDF", as per table this is my hallway which is the easiest to work with for me right now. Also I used "watch 50 BDF", as I get updates after changing things to correlate easier.
Ok. So lets start:
Switching to automatic mode:(21°): `gateway(0x50) B all(0x00), ?(0x0BDF), data: 00 E1 3D 2A`
Setting temperature to 18°:` gateway(0x50) B all(0x00), ?(0x0BDF), data: 00 E2 3D 24`
switching to manual mode 20°: `gateway(0x50) B all(0x00), ?(0x0BDF), data: 00 E2 3C 29`
setting temp to 15,5°: ` gateway(0x50) B all(0x00), ?(0x0BDF), data: 00 E3 3C 1F`
switching to "off": no telegram is arriving on 50 BDF, and the next one I receive(not in the same close amount of time as before) is: `gateway(0x50) B all(0x00), ?(0x0BDF), data: 00 E3 3D 2A
`
Doing "watch F7" shows me this when switching modes on BDF:
switching to auto: `gateway(0x50) B all(0x00), MenuConfig(0xF7), data: FF 0A B7 D3`
switching to manual: `gateway(0x50) B all(0x00), MenuConfig(0xF7), data: FF 0A B7 D9`
switching to off: `gateway(0x50) B all(0x00), MenuConfig(0xF7), data: FF 0A B7 D1
`
and this is the same sequence on BE2 (kitchen):
```
gateway(0x50) B all(0x00), MenuConfig(0xF7), data: FF 0A BA D3
gateway(0x50) B all(0x00), MenuConfig(0xF7), data: FF 0A BA D9
gateway(0x50) B all(0x00), MenuConfig(0xF7), data: FF 0A BA D1
```

read 50 1230: (on "watch 50 1230" nothing happens , also when playing around with modes)
```
2025-09-28T21:30:25.300 N 320: [emsesp] me(0x0B) R gateway(0x50), ?(0x1230), length: 25
2025-09-28T21:30:25.416 N 321: [emsesp] gateway(0x50) W me(0x0B), ?(0x1230), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

"watch unknown"
setting kitchen (BE2) to auto:
```
gateway(0x50) B all(0x00), MenuConfig(0xF7), data: FF 0A BA D3
gateway(0x50) B all(0x00), ?(0x0BBA), data: 00
```

setting kitchen to manual:
```
gateway(0x50) B all(0x00), MenuConfig(0xF7), data: FF 0A BA D9
gateway(0x50) B all(0x00), ?(0x0BBA), data: 01
```

setting kitchen to off:
```
gateway(0x50) B all(0x00), MenuConfig(0xF7), data: FF 0A BA D1
gateway(0x50) B all(0x00), ?(0x0BBA), data: 03
```

I tested the same with the main hall (BDF):
auto:
```
gateway(0x50) B all(0x00), MenuConfig(0xF7), data: FF 0A B7 D3
gateway(0x50) B all(0x00), ?(0x0BB7), data: 00
```

manual:
```
gateway(0x50) B all(0x00), MenuConfig(0xF7), data: FF 0A B7 D9
gateway(0x50) B all(0x00), ?(0x0BDF), data: 00 E5 3B 1F
gateway(0x50) B all(0x00), ?(0x0BB7), data: 01
```

off:
```
gateway(0x50) B all(0x00), MenuConfig(0xF7), data: FF 0A B7 D1
gateway(0x50) B all(0x00), ?(0x0BB7), data: 03
```

> You can try to add some custom entities for device 0x50, first sensor
> 
>     * mode: tele 0xBB5 offset 0, UINT8, writeable
> 
>     * setpoint auto: tele 0xBB5, offset 1, UINT8, writeable
> 
>     * setpoint manual: tele 0xBB5, offset 3, UINT8, writeable
> 
>     * roomtemp, tele 0xBDD, offset 0, UINT16
> 
> 
> I suggest to map the MX400 on 0x50 as device `connect`, like other wireless controllers from bosch. The add the thermostats tagged to this device. For first test i will use the hs1-hs16 tags, we already have this for heatingsources, thisdoes not conflict and we know the tags are working. Later we can add new tag names, eg. room1.... or th1

continued...

MichaelDvP, 2025-09-29:
You could test https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest with a first implementation.

Roxxor91, 2025-10-16:
> > To be clear on the HA stuff: EMS-ESP, BC400 and heat pump are working and reporting as intendend. Only the newly added MX400 shows unavailable entities. How to debug?
> 
> Maybe something in HA, the old entities were registered with `obj_id`, now we have changed to `def_ent_id`, see [#2640](https://github.com/emsesp/EMS-ESP32/issues/2640). What HA version do you use? Just for testing i have added back the `obj_id` to my dev build, try it. (test build without obj_id, but up to date) If you update to HA 2025.10 you should see the `deprecated option` warning again in HA.

I am using HA 2025.10.2. And it is very weird. The MX400 now doesn't appear again. And also after removing die BC400, heat pump and EMS-ESP reappear flawlessly, but including the rename of the devices. So it doesnt really start from scratch, also after removing them. The obj_id error I also just see from my (not yet updated) Valetudo vacuum. Not sure how I could perform a really fresh start for the HA autodiscovery. Talking of auto discovery: Is this correct?

[image]

> > [@proddy](https://github.com/proddy) Coulld you please check the attached mqtt config for HA: Fixed the climate in my last dev build, but not sure what could be wrong with other entities. The sensor/number configs looks ok to me. (I have only created entity-id format: multi-instance-short-name). [mqtt_HA.txt](https://github.com/user-attachments/files/22891871/mqtt_HA.txt)
> 
> The HA topics look correct. Let's see which version of HA is being used and [@Roxxor91](https://github.com/Roxxor91) would need to look at the HA logs and perhaps add verbose logging to the HA add-on to get more details. I'm not hearing of any other users facing this issue so I suspect its a config somewhere.

Heat pump, BC400 and EMS-ESP work without any hunch. They work perfectly in HA. Its only the MX400 devices. Maybe at some point it got an "incorrect" auto discovery while testing all the builds, and now is somehow stuck and doesnt really rediscover from scratch? Also after removing!
5 minutes ago I changed the base topic to ems-esp-test. But now the old entities are disconnected (as expected), but up to now, no new devices popped up.

MichaelDvP, 2025-10-16:
Try : Uncheck `[ ] Aktiviere Mqtt Discovery`, click Apply settings (but leave option mqtt enabled checked). Wait 5 minutes, Enable Mqtt discovery again. This will clear old  mqtt -HA configs on the broker and generate new ones.

proddy, 2025-10-20:
just merged it @mranybody0

MichaelDvP, 2025-10-20:
> just merged it @mranybody0

I have not count up the version, so binaries are not created. I do not follow the update rules strictly on my site and build the binaries manualy.

mranybody0, 2025-10-21:
thank you, that did the trick - working perfectly now